robototes/Reefscape2025 #37: Driving improvements

KangarooKoala, 2025-02-21:
* [ ] Add slow mode (while button is held)
* [ ] Apply deadband (via [`MathUtil.applyDeadband(double, double)`](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/math/MathUtil.html#applyDeadband(double,double))) on each axis separately
* [ ] Activate acceleration limit
* [ ] Apply deadband to translation magnitude

Activating the acceleration limit depends on applying the deadband ourselves so that the deadband is done before the acceleration limit. Otherwise, these can be done in any order (though it is recommended to apply deadband on the axes separately before trying to apply deadband to the translation magnitude).

iamawesomecat, 2025-03-03:
Are we still adding the deadband to translation magnitute? or did we already do that?

KangarooKoala, 2025-03-03:
After discussion with James, the deadband is supposed to be unnoticeable to the driver anyways, so it's fine to apply the deadband to each axis separately. There are also effects on the direction of motion from applying the deadband to each axis separately (The deadband region is a thin plus, not a circle), but they should be negligible.