robototes/Reefscape2025 #52: Spinny claw motor keeps running after coral is held, using 30-40 amps continually

jamesdooley4, 2025-03-04:
Looking at logs for drive practice, I noticed that motor 40 (spinny claw) draw between 40 and -47 amps, doesn't really change position (so is likely stalled) and has a continual amount of small jitter. It also an input voltage of around 6.5-8 volts during this time, even though other motors (I picked a drive motor) report about 11.5v of input voltage. It should detect that a coral is being held (or that the motor is enabled but not moving) and then stop itself instead of continuing to try and run & waste power.

[image]

KangarooKoala, 2025-03-04:
Probably fixed by #47, but we should have people check logs today to make sure.

KangarooKoala, 2025-04-01:
Fixed by #112.

closed 2025-04-01 by KangarooKoala