robototes/Reefscape2025 #64: Reduce LED power draw while disabled

jamesdooley4, 2025-03-17:
When the robot is disabled, the LEDs should reduce power usage in order to conserve battery life. If a team has control system issues, sometimes our robot may sit on the field for a very long time before a match and it's important to conserver power during that time. So, something that goes into a 'sleep mode' after some preset amount of time while disabled might be an option.

jamesdooley4, 2025-03-17:
To measure power usage, consider combining this with #65

KangarooKoala, 2025-03-17:
After discussions with drive team, the animations should not be dimmed while enabled, but we can reduce power draw in the following ways:
* The initial color set to all white (in the constructor) can be further dimmed
* The command that runs when the elevator is not zeroed can be dimmed a little (like what is done on https://github.com/robototes/Reefscape2025/tree/Elevator-LEDs2) and should blink every two seconds or so (use [`.blink(Time onTime, Time offTime)`](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/wpilibj/LEDPattern.html) so that the LED is on for a short time and the total time is two seconds)
* The command that runs when the elevator is zeroed can turn off after a second (ideally fading to zero after that, but because the LEDPattern API doesn't let us specify the initial phase of the animation, for now we'll go straight from on to off)
* Bind the elevator not zeroed and elevator zeroed animations to run when disabled (to overwrite the last set color)

If we use Elevator-LED2, we should make sure to restore the brightness of the LED animations that run while we're enabled (climb, automatic intake, and superstructure).

jamesdooley4, 2025-03-19:
With #76, the LEDs in all white take 2.3 amps. In all red, they take about 1 amp.
The LEDs are on PDH channel 3 along with the power going to the two CAN loops for the sensors and CANDi on them, which take about .3Amp (measured by unplugging the LEDs)

RobototesProgrammers, 2025-04-18:
Closed in #101. (We don't overwrite the last set color upon disable when elevator is zeroed, but that's semi-intentional so we still have that information)

closed 2025-04-18 by RobototesProgrammers