robototes/Reefscape2025 #66: Prevent presets from getting stuck waiting for arm to finish rotating

jamesdooley4, 2025-03-17:
During drive practice on 3/16 (logs in 
[March16practice.zip](https://bsd405.sharepoint.com/:u:/r/sites/SammamishRoboticsTeam2412/Shared%20Documents/Software%20Team/2025%20Practice%20logs/March16practice.zip?csf=1&web=1&e=Kyd8A0?csf=1&web=1&e=Kyd8A0) ) there were a number of times where pressing the 'Score' control button didn't do anything. One theory from @KangarooKoala is that the arm doesn't get to its final rotation position so the command to rotate to that angle isn't finished, preventing the Score command from being able to run.

KangarooKoala, 2025-03-17:
On the theory that the arm is getting stuck, we can make the following changes:
* When we start the preset, instead of using something like `parallel(spinnyClawIn, elevatorToHeight, armPivotStraightUp)`, we should use `parallel(spinnyClawIn, elevatorToHeight.deadlineFor(armPivotStraightUp))`. The `commandA.deadlineFor(commandB)` means "run A and B at the same time, and end when A ends, interrupting B if it isn't done yet". We only care about moving arm straight up so that it stays out of the way when traveling up, so when we can move on once the elevator reaches the top.
* For L2 and L3, when we wait to move on in the preset, instead of having `armPivotPreScore` and `Commands.waitUntil(score)` in a sequence, we can experiment with using `armPivotPreScore.withDeadline(Commands.waitUntil(score))`. `commandA.withDeadline(commandB)` is similar to `deadlineFor()` but with the opposite direction (so it means "run A and B at the same time, and end when B ends, interrupting A if it isn't done yet"). Notably, unlike using `.until()` with a BooleanSupplier, `withDeadline()` will also wait for `commandB` to end even if `commandA` is done early. We will need to test how this handles early button presses, which is how the current auto setup works. (Long-term, we'd want auto to set the score condition to be when we're at the end of the path, but that might not be done in time for WAAHS)
* When we have the robot, we can try tuning the arm control gains (particularly P and D) to make the arm move more quickly to the target position.
* As a temporary solution (if the above doesn't work), we can increase the arm pivot angle tolerance because we don't really care about going to a precise angle (just moving through a particular angle when scoring).

KangarooKoala, 2025-03-24:
Completed in #62 and #82.

closed 2025-03-24 by KangarooKoala