robototes/Reefscape2025 #70: Calibrate cameras for all regions of the field

jamesdooley4, 2025-03-17:
During drive practice on 3/16 (logs in 
[March16practice.zip](https://bsd405.sharepoint.com/:u:/r/sites/SammamishRoboticsTeam2412/Shared%20Documents/Software%20Team/2025%20Practice%20logs/March16practice.zip?csf=1&web=1&e=Kyd8A0?csf=1&web=1&e=Kyd8A0) ) there were a number of times where the vision values were either wildly off (robot 20 feet away, or 8 feet high and tilted) as well as times where the left & right cameras disagreed on the robot position. This was especially noticeable in the FRC_20250317_002619.wpilog file when the robot was on the autonomous start line & its position 'vibrated'. This is likely one major cause of why any of the autonomous routines run from that starting location failed to score correctly on the reef poles.
The camera positions & rotations need to be better calibrated so they work from next to the reef up to ~7 feet away from the reef, and ~7 feet away any of the other AprilTags that are visible.
One way to do this is physically moving the robot to a known location relative to each of those AprilTags, and then adjusting the camera position (x, y, z, roll, pitch, yaw) so the vision-provided position matches the physically measured position of the robot. Another way is to keep the robot in a single spot & physically move an AprilTag to a location with a known physical offset from the robot, and then adjust the camera position (x, y, z, roll, pitch, yaw) so the vision-provided position matches the physically measured position of the robot.

KangarooKoala, 2025-03-17:
On 3/15 it looked like the camera positions were accurate, but we can always double-check. More research into why the position is vibrating is also important- I suspect either we get a bad tag in view or the wrong tag direction gets picked from the 2d->3d ambiguity. No matter what, we should recreate the scenario and look at the camera streams to see what's happening.

A related issue I noticed when going through the changes in #63 is that we commented out the listener for updates in the left camera. The right listener causes both cameras to be polled for updates, so we still get results from the left camera, but only when we also get results from the right camera.

koolpoolo, 2025-03-17:
Yeah and as far as I know if vision says the bot is super high then it's reading a bad tag (just a note)

jamesdooley4, 2025-03-17:
FYI that drive team adjusted the AprilTag camera resolution to be higher than 800x600 and it now has a visibly larger FoV

RobototesProgrammers, 2025-03-18:
calibrated the new resolution but it runs at 14 fps and 80ms latency

RobototesProgrammers, 2025-03-18:
while the old resolution runs at 30 fps and 30 ms latency

KangarooKoala, 2025-03-18:
> calibrated the new resolution but it runs at 14 fps and 80ms latency

> while the old resolution runs at 30 fps and 30 ms latency

Thanks for tracking the performance difference! 14 fps with 80ms latency is still fine. Since vision just needs to correct odometry (which should be able to stay accurate for the 33ms between frames at 30 fps), 14 FPS is plenty. 80ms latency is also fine- Odometry should be able to stay accurate for at least 80ms anyways. (For reference, at WASAM we essentially didn't have vision and odometry was quite good for at least 5 seconds of the auto, which is over 50 times longer than 80ms.)

markpete, 2025-03-25:
Some of the teams with better alignment only use 1 camera at a time (even though they had 4 mounted on their bot).  They only use the left front when moving forward right and only the right front when moving forward left.  Worth playing with to see if we can get better results in this way versus always on 2 cameras.  Will require testing.

closed 2025-03-31 by koolpoolo