robototes/Reefscape2025 #81: Auto from middle tries to score while elevator is moving up, resulting in coral not scoring or being hit by reef poles on the way up

jamesdooley4, 2025-03-23:
In Q58 and Q72 of WAAHS the 'score' condition appeared to evaluate to true while the elevator was being raised, causing the arm target position to jump to 0.0625 (CORAL_PRESET_PRE_L4) while the elevator was still going up. In both the arm then was commanded to go to CORAL_PRESET_DOWN (-0.25) later while the elevator was still going up. In Q58 the elevator then went back down due to a different issue (Coral briefly appeared to be dropped) that's been fixed. In Q72 the elevator stayed up and nothing more happened during auto.

KangarooKoala, 2025-03-24:
It would be nice to log `readyToScore()` to make it easier to determine why the superstructure cycles through the presets.

A good fix would be to change the superstructure logic so that in auto, the score condition only interrupts arm movement, not elevator movement. (We may want to change the logic to only interrupt arm movement in all periods of the match)

KangarooKoala, 2025-04-02:
We now have logic so that `readyToScore()` only triggers if the robot is within a few centimeters of the branch pose. (We also have logging of the separate parts of the `readyToScore()` condition!)

We have also changed superstructure logic so that the score condition only interrupts arm movement, not elevator movement.

closed 2025-04-02 by KangarooKoala