robototes/Reefscape2025 #105: Superstructure fixups

KangarooKoala, 2025-03-26:
Thought of some more things:
* [ ] Apply the `runOnce(() -> CommandScheduler.getInstance().schedule(...))` fix to the intake binding as well.
* [ ] Combine the branchHeight and algaeIntakeHeight assignments into one `runOnce()` command instead of two. The combined command should combine the names of both commands. (e.g., "coral level 4, algae level 3-4")
* [ ] Remove or update the comment on the claw spit binding (driver right bumper, in `configureSpinnyClawBindings()`
* [ ] Remove old `SuperStructure.coralLevelFour()` comment
* [ ] Move `SuperStructure.java` out of the subsystems folder. (It's not a subsystem, it's a combination of subsystems)

KangarooKoala, 2025-03-31:
L1 worked fairly decently, but we forgot to actually move SuperStructure.java.

iamawesomecat, 2025-04-01:
Good to close?

KangarooKoala, 2025-04-01:
Moved the last item to #116.

closed 2025-04-01 by KangarooKoala