robototes/Reefscape2025 #108: Climb fixes

KangarooKoala, 2025-03-28:
Missed a few things earlier:
* Run the motor in either direction depending on the direction funnel needs to move. Right now it only bases the direction and speed based on the target position, regardless of where the funnel is right now.
* Prevent climb from running immediately upon reenable. We should tell the motors to stop when we enter disabled motor to clear any previous setpoint, like how [elevator does it](https://github.com/robototes/Reefscape2025/blob/ad2de345cd312866bdf0e494c65b643e83374e30/src/main/java/frc/robot/Controls.java#L368).
* Delete commented out code in `periodic()`.
* Update inTolerance and moveComplete in `periodic()`, so that updates always happen and all that the default command does is check the variables and set the speed accordingly.

We'll also need to see whether the presets work.

iamawesomecat, 2025-03-28:
Ask drive team to try and climb today?

KangarooKoala, 2025-03-28:
If there's dead time, maybe.

Drive team has the highest priority since this is their time. Since this won't really have a major impact on the presets working or not working, I don't think this would significantly improve drive practice, so the way to maximize drive practice quality would be to not take up any drive practice time with this.

iamawesomecat, 2025-03-28:
Yeah, I mainly meant to practice climb alignment, as climbing is something they should try anyway. But if it isnt too beneficial then yeah they shouldnt

PR #110 (2025-03-29): Various climb fixes 3/28
fixes #108
closed 2025-03-29 by KangarooKoala