robototes/Reefscape2025 #117: Document camera ports

KangarooKoala, 2025-04-01:
PhotonVision uses the USB ports on the Orange Pi to match the cameras. If we plug the cameras into the wrong port, then it will not be able to properly detect which camera is which. We should document the following which port each camera should be plugged into.

Wherever we document the camera ports should be easily accessible during a competition. Some potential spots are on the backs of the cameras or on the Orange Pi case.

KangarooKoala, 2025-04-02:
We need to find a better spot to document this, but:
* Port 0 is the top-right port when looking into the USB ports.
* Port 1 is the bottom-right port when looking into the USB ports.
* Port 2 is the left port when looking into the USB ports.
* The right camera needs to go into port 0 (the top-right port).
* The left camera needs to go into port 2 (the left port).

TrisDooley, 2025-08-19:
Is this still true? They can get labeled at drive practice on Wednesday.

koolpoolo, 2025-08-25:
we can do it if we want to but yes

koolpoolo, 2025-08-25:
@TrisDooley

TrisDooley, 2025-09-29:
This happened

closed 2025-09-29 by TrisDooley