robototes/Reefscape2025 #127: Net auto align

KangarooKoala, 2025-04-05:
This would improve consistency with net scoring, which would grant extra flexibility to complement any alliance we end up on.

Net auto align process, in psuedo-code:
```java
sequence(
  driveToBlackLine(), // Use PID control on the robot X position and angle
  slowlyDriveForward()
    .withDeadline(
      superstructure.algaeNetScore(() -> atScoringXPosition())
```

In progress on #126, but still needs the rest of the sequence. I'd recommend using `new BargeAlign(...)` instead of `driveToBlackLine()` and defining `BargeAlign.bargeAlign()` factory method with the sequence described above.

It should also be relatively simple to rewrite the `BargeAlign` class command to be defined by a factory method, but continuing to use a class command also works.

KangarooKoala, 2025-04-08:
Logic has been tested on #126 (minus different combinations of alliance color and robot position), and precise positions still need to be tested.

RobototesProgrammers, 2025-04-18:
Closed in #126.

closed 2025-04-18 by RobototesProgrammers