robototes/Reefscape2025 #128: Clean up auto align logic

KangarooKoala, 2025-04-05:
It works well, but now that we're no longer in the middle of an event, we can clean up the logic and organization. This way, we can more easily reuse and adapt this in future years.

Ways to clean things up:
* Remove unused `AutoAlign.java` logic. (`MaxAcceleration`, `MaxAngularAcceleration`, branch pose constants, `autoPathAlign()`, `angle` swerve request, `aim()`, `readyToScore()`, `isCloseEnough()`, and `getClosestBranch()`)
* Move `AutoAlignTwo` logic into `AutoAlign.java`.
* Remove "two" from new auto align logic names. (There's only one now that we removed the old logic)
* Make the auto align command constructor private. (It now lives inside the same class as the factory method, and making it private ensures that everything goes through the factory method)
* Remove the defer logic in the factory method. (We don't need to recreate the command every time it starts, since it doesn't have anything at construction time that needs to be rerun)
* Move `blueBranchPoses` and `redBranchPoses` below the branch pose variables and initialize them in the member declaration instead of using the `static {}` initializer block.
* Move non-static variables to right above the constructor. (Makes it easier to separate static and non-static stuff)
* Remove unused `getNearestTag()`? Not sure if we need it for something that'll be added later, but right now it doesn't do anything.
* Use `.withForwardPerspective(ForwardPerspectiveValue.BlueAlliance)` and remove the logic to flip the x and y speeds when on the red alliance. Telling something not to flip is cleaner than flipping something twice.
* Remove check for proximity to the branch in `execute()`. The same logic already happens in `isFinished()` as well, and after `isFinished()` returns true, `execute()` is no longer run. Unduplicating code also makes it easier to make changes later on.

KangarooKoala, 2025-04-08:
In progress on https://github.com/robototes/Reefscape2025/tree/AutoALignCleanup. As of https://github.com/robototes/Reefscape2025/commit/f708dd0c2465663d2aeea41eb1525bf1d0d01e6f, we just need to do the forward perspective logic and remove the proximity check in `execute()` (which we've done already for barge align).

jamesdooley4, 2025-04-08:
> Remove unused getNearestTag()? Not sure if we need it for something that'll be added later, but right now it doesn't do anything.

getNearestTag was intended to be used to signal the driver if the AprilTag being driven towards wasn't visible from the PhotonVision cameras. 'Bad things' (such as trying to drive across the field) happen when spurious tags are seen and the vision-based position is wildly off and auto-align is invoked.

Jetblackdragon, 2025-04-08:
All fixes should be ready to be merged. getNearestTag and april tag lists have been removed and now the rumble just happens if we arent getting vision detections. The changes from this issue were done in autoaligncleanup and the rumble stuff is being done in auto-align-rumble

closed 2025-04-08 by Jetblackdragon