robototes/Reefscape2025 #130: Clean up climb target selection logic

KangarooKoala, 2025-04-05:
Right now, we duplicate the logic for changing the target position in a couple of different locations.

One way to fix this to define a `private void setTargetPos(TargetPositions newTargetPos)` which handles updating `targetPos`, `maxTargetPos`, `minTargetPos`, `holdSpeed`, and `moveComplete` depending on the new target position. Then `advanceClimbTarget()` and `stow()` just need to call `setTargetPos()` with the right target position, and the logic is only defined in one spot that gets reused.

Separately, we should also add a name to the `stow()` command factory.

TrisDooley, 2025-04-07:
Can this come as a combo with having one button cycle between "latch" and "coral intake" positions, and a second button go to "climb"? This reduces the amount of thinking required during the most time-limited (and stressful) part of the match. Potentially linking driver "score" while the funnel is out, and the bot doesn't have a game piece, to going to "climb" position.

Examples
"coral" -> B1 -> "latch" -> B1 -> "coral"
"latch" -> B2 -> "climb" -> B2 -> "climb" (still)
"latch" -> B2 -> "climb" -> B1 -> "latch"
"climb" -> B1 -> "latch" -> B1 -> "coral" -> B2 -> "climb"
"coral" -> B2 -> "climb"

KangarooKoala, 2025-04-07:
To prevent trying to go to stow position while climbing, the current setup is that operator start cycles between climb out and climbed (which I think you call "climb" and "latch") and operator right trigger unconditionally goes to stowed (or "coral intake"). That way, the climbing sequence can never accidentally go into the stowed/coral intake position, which we know causes problems.

Ideally we'd have three separate buttons so that there's zero state tracking needed... I think operator pov right and pov up are free? Those might also be used for some ground intake features though...

Also, just to avoid any possibility of confusion about climb out vs climbed, the climbing sequence is going to climb out (funnel mostly vertical as we latch the cage) and then going to climbed (funnel mostly horizontal to lift us off the ground). I'm not sure which you call climb and which you call latch, but hopefully this description will let us figure it out.

TrisDooley, 2025-04-07:
I agree 3 buttons would be best, maybe one of them can be combined with something else? Since coral mode already sets the robot up for coral intaking, maybe that can also send funnel to "stowed". Maybe operator sending the robot to pre-intake (regardless of mode) brings the funnel to "stowed" as well.

"coral" = "stowed" = "coral intake" = ~30 degrees below horizon
"latch" = "climb out" = ~80 degrees above horizon
"climb" = "climbed" = ~30 degrees above horizon

KangarooKoala, 2025-04-08:
We ended up making operator left trigger (coral mode select) go to stowed/coral intake, right trigger go to climb out/latch, and start button go to climbed/climb.

closed 2025-04-08 by KangarooKoala