robototes/Reefscape2025 #132: Sort out default and do nothing autos

KangarooKoala, 2025-04-05:
Before #124, the default was labeled as "do nothing" when it just drove forward. To make it easier to test operations in auto that happen regardless of the path (e.g., raising the climb funnel), we added an actually do nothing auto which shows up in every configuration (starting position and piece number).

We should sort out which one should be the default path for every configuration. We should also sort out whether the other should also show up in every configuration or just in particular configurations.

Jetblackdragon, 2025-04-08:
I made a commit on autoaligncleanup that now has an actual do-nothing path and it will just always be the default path.

KangarooKoala, 2025-04-08:
We'll close this when the auto align cleanup PR is merged. By the way, if you edit the PR description to say "Closes #132", then merging the PR will automatically close this issue.

RobototesProgrammers, 2025-04-18:
Closed in #139.

closed 2025-04-18 by RobototesProgrammers